uStepper/uStepper #2: Missing defines in library

richardw347, 2016-04-05:
The library refers to constants for pins DIR, STEP, ENA which are never defined as such the library and examples do not compile.
```
pinMode(DIR, OUTPUT);
pinMode(STEP, OUTPUT);
pinMode(ENA, OUTPUT);	
```

uStepper, 2016-04-05:
Hi Richard 

IT sounds like you only installed the library in arduino, and not the hardware part. The three constants you refer to are not defines in the library, but constants in the hardware pin mapping file. 
Did you follow the install guide on the front page of our github page? 

Best regards, 
Thomas Olsen

richardw347, 2016-04-05:
Hi Thomas,

Thanks I didn't realise you has a custom board def for arduino, just downloaded the library directly!

closed 2016-04-05 by richardw347

uStepper, 2016-04-05:
You are very welcome! 
Feel free to contact me again at Thomas@ustepper.com if you run into further trouble

Best regards 
Thomas Olsen