uStepper/uStepper #3: bug uStepper.cpp

python36, 2016-04-28:
`if(state != STOP)` to `if(this->state != STOP)`

and after

`this->direction = dir;`

must be

`digitalWrite(DIR, dir);`

uStepper, 2016-04-28:
Hi,

Thank you for your feedback.

However I am not sure i agree with you. Can you be a bit more specific?
The lines you are referring to is used a few places in the code, do you want all of them to be changed or just one place?
Also, can you explain what bug this change will fix?

Best Regards,
Thomas Olsen

python36, 2016-04-28:
Sorry, I badly speak English, so I'll show the code that is hard to work without any changes I wrote.   
 `if (abs(value - old_value) > 24)
      {
        old_value = value;
        stepper.setMaxVelocity((SMAXSPEED - SMINSPEED) / (MAXTIME - MINTIME) * abs(value - ((MAXTIME - MINTIME) / 2 + MINTIME ))*10);
      }
    
    if (value > 1610 )
    {
      stepper.runContinous(CW);
    }
    if (value < 1590)
    {
      stepper.runContinous(CCW);
    }
    if (status != -1 && value > 1590 && value < 1610)
    {
      status = -1;
      stepper.hardStop(SOFT);
    }`
value - RC signal

uStepper, 2016-04-28:
I still don't quite understand why the changes you suggest is necessary. However, the code snippet you just posted result in quite a performance degradation. Every time runContinous() function is called, the timer which calculates the stepping algorithm is stopped, and every variable this interrupt routine uses is recalculated, which is actually quite computationally heavy. runContinous is not designed to be called as often as you do in the above code snippet.
Instead you should use a variable, to keep track of which direction the motor should rotate, and only call runContinous() function if this direction changes. for example:

```
uint8_t motorDir = 2;
..
..
..

      if (abs(value - old_value) > 24)
      {
        old_value = value;
        stepper.setMaxVelocity((SMAXSPEED - SMINSPEED) / (MAXTIME - MINTIME) * abs(value - ((MAXTIME - MINTIME) / 2 + MINTIME ))*10);

         if(value > 1610)
         {
           motorDir = CW;
         }
         else if(value < 1590
         {
           motorDir = CCW;
         }
         else 
         {
           motorDir = 2;
         }
       }

       if ( ( motorDir == 2 ) && stepper.getMotorState())
       {
         stepper.hardStop(SOFT);
       }
      
        else if ( ( motorDir != (uint8_t)stepper.getCurrentDirection() ) || ( stepper.getMotorState() == 0 ) )
        {
          stepper.runContinous((bool)motorDir);
        }
```

The code above will be far more efficient than the one you posted. I assume you only use "status" variable to make sure the "hardStop()" function is not called in every single loop? Therefore i have removed it in the above, and just used the getMotorState() function to check if the motor is running or not.
Even though i don't know exactly what your specific problem is, i would suspect that the increased efficiency of the code snippet i just posted will fix your problem without changing the lines you suggested in your first post. 

You should note that i have NOT tested the code snippet i just wrote but it should do the same as the one you wrote.

the reason why i don't agree to make your suggested changes is the following:
The first change you request is actually not a change, since the two lines are equivalent.
The second change you request is going to be counter productive, since the motor should ramp down in speed before changing direction. Therefore if i just change the state of the direction pin, the motor will instantly try to rotate at unchanged speed but in the different direction and then ramp down in speed and ramp up again. This will result in a strange speed curve and possible make the motor stall. Therefore i change the direction variable instead and let the stepper algorithm handle the direction changing.

Best Regards,
Thomas Olsen

closed 2016-05-27 by uStepper