uStepper/uStepper #4: Question about AS5600 I

Dark-Guan, 2016-06-11:


uStepper, 2016-06-11:
Hi Dark-Guan.

Do you mean that the high and low byte of the angle read from the encoder does not match within the sample period?

Can you elaborate on this?

Dark-Guan, 2016-06-11:
Hi uSteper
Thanks for your replay.
I find this problem ,And I give up the AS5600 .

Dark-Guan, 2016-06-11:
I have compared all the product of AMS Magnetic Position Sensors, It seems the ones support digital angle (interface) or AB(ABI) are more suitable for the accurate position detecting for motor.

uStepper, 2016-06-11:
We dont do it like this in the new version of the library, since we had to write a whole new I2C class to replace the one in arduino, so what you are describing cannot happen in the new library.

i have never experienced this issue, with either the old or the new code.

Have you experienced this problem, or are you just afraid it might happen?

Dark-Guan, 2016-06-11:
Yse,I have experenced this problem too.
It seems AS5600 does not lock the Angle Register (either the high and the low),when sends the the data,If it is ture,there is no way to keep the position messsage right  all the time ,espercially when the motor is running,I build a close-loop controlled motor driver use this Chip,but the error is too bigger.
If you have solved the problem?

Dark-Guan, 2016-06-11:
I will try AS5601 and AS5145B ,I think they will be better.

uStepper, 2016-06-11:
okay. However this problem can not happen if the two bytes are read back to back, meaning that instead of closing the I2C bus after the first byte has been read, you just read the next byte straight away.

This way both bytes has been transferred over the I2C port faster than the internal sampling frequency of 6.6kHz inside the AS5600 chip. (Here i assume you use 400kHz or higher I2C clock)

Dark-Guan, 2016-06-11:
Yes,I am reading your newest code.
It seems I should read the As5600 datasheet again.
Thanks for your replay and advice very much,I will try your method

closed 2016-06-11 by uStepper

Dark-Guan, 2016-06-11:
I have a question again,Can you tell me where do you find the message of the internal sampling frequency of 6.6kHz inside the AS5600 chip.

uStepper, 2016-06-12:
It is listed in the datasheet of the AS5600

Dark-Guan, 2016-06-12:
I test your method today,There is a little hop still ,no mater when disable the timer1 or not during read the angle.
I want to report this to you.
Thanks very much for your reply!

IanSC, 2022-01-02:
try reading the 2 bytes together:
```
word AMS_5600::readTwoBytes(int addr)
{
  Wire.beginTransmission(_ams5600_Address);
  Wire.write(addr);
  Wire.endTransmission();
  Wire.requestFrom(_ams5600_Address, 2);
  while (Wire.available() < 2);
  
  int high = Wire.read();
  int low = Wire.read();
  return ( high << 8 ) | low;
}
```