uStepper/uStepper #5: uStepper wailing

chocol4te, 2016-08-11:
This may be my own fault, but I am having an issue where I have a uStepper motor and two end-stops. I have attached the end stops as hardware interrupts to call .hardStop. When this occurs during a .moveSteps, the motor stops but then "wails" very loudly. No other commands will work after this. Is there any other way I can ensure the motor does not go out of range or is there a fix to this issue? Thanks

uStepper, 2016-08-11:
Hi,

I just just tried to recreate the problem you describe. I am however, not able to do this. 
When using hardStop(HARD), the motor will stop without deceleration and make the motor block. This will result in a vibrating sound in the motor (caused by the motor windings being excited), although it will not block communication or any other functions in the board.

Can send me your sketch so i can try exactly the same setup as you?
Also, which version of the library are you using, and which version of the arduino IDE are you using?

Best Regards,
Thomas Olsen

chocol4te, 2016-08-12:
Thank you very much for your reply! I sincerely apologise, I found the bug in my code which unintentionally sent the stop command repeatedly. Again, my apologies.

closed 2016-08-12 by chocol4te

uStepper, 2016-08-12:
I am glad you found the problem. 

Best Regards,
Thomas Olsen