uStepper/uStepper #7: SOFT holdMode does not function when using closed loop control

chrisbeardy, 2017-08-18:
after setting up closed loop control, the soft options for holdmode no longer free up the motor to freely rotate, it constantly tries to correct its position, in order to "disable" the power stage need to recall setup() with no parameters the softStop(SOFT) : see

#include <uStepper.h>

#define maxacceleration 1500// steps/s^2
#define maxvelocity 3200// steps/s
uStepper stepper(maxacceleration,maxvelocity);//instanstiate
String command;

void setup() 
{
  //movement
  stepper.setup(PID,SIXTEEN,10,5,1.0,0.02,0.006); 
  //comm
  Serial.begin(9600); 
}

void serialEvent() 
{
  command = "";
  while (Serial.available())
  {
    // get the new byte:
    char inChar = char(Serial.read());
    // add it to the inputString:
    command += inChar;
    delay(5);
  }
}

void loop() 
{  
    if (command.substring(0,2)=="01") //move cont 
    {
      stepper.runContinous(CW);
      command = "";
    }
    else if(command.substring(0,2)=="06") //soft stop of the motor mode
    {
      stepper.softStop(SOFT);
      command = "";
    }
    else if(command.substring(0,2)=="07") //hard stop of the motor mode
    {
      stepper.hardStop(SOFT);
      command = "";
    }
}

uStepper, 2017-08-20:
Hi!
The idea is to not allow position errors when the loop is closed, i.e. the PID is always running. We have however made an adjustment just 3 weeks ago, which allows you to disable the PID temporarily. Have a look in the documentation under setup [HERE](http://ustepper.com/docs/html/classuStepper.html#a9efdd422fe5981a86929c274bf671807)

When you need to "pause" the PID you call the setup again like this:
stepper.setup(PID,SIXTEEN,100000,50000,1.0,0.02,0.006,0);
You now have a lot of hysteresis to move stuff around, while still keeping track of position. The last digit in the setup is the NEW thing. 0 = "do not reset the position with this call to setup", 1 = "set the current position to the new 0 with this call"
So when you have moved things manually, you simply reinitialise the control with:
stepper.setup(PID,SIXTEEN,10,5,1.0,0.02,0.006,0);
So now with the hysteresis you want and still the 0 in the end to not reset the position counter.

Please get back to me with the result when you have had a chance to check if it works!

Best regards,
Mogens

chrisbeardy, 2017-08-29:
HI, 

this works nicely for my application however there are some ideas that I think could be implemented to make it work more smootly:

- A layer of abstraction would be nice so that the user can just call motorenable(on/off) or disablepid() etc
- it is very common in engineering applications to want to disable the motors so that they will freerun so i think making this a permanent thought out feature would be useful
- the current drawback is that as soon as you reach the extended hysteresis the motor will try to move back to its set position which if the motors being used in let say a nip roller scenario (what i'm using it in), it could freewheel for a while, luckily in my application its only a few turns so it works ok.  
- it would be good if disabling the motor would be quicker, so you could have a smooth transfer from controlled movement to freewheeling

let me know what you think?

uStepper, 2017-09-04:
Hi Chris,
Actually you can do this: stepper.setup(NORMAL,SIXTEEN,10,5,1.0,0.02,0.006,0);

But having a dedicated function for pausing the PID sounds like a good idea. We will get it on our todo list for the coming updates.

Thanks for the feedback!

Best regards,
Mogens

closed 2018-01-24 by uStepper

BeebBenjamin, 2018-11-03:
Do you have plans to implement this?  Or have you already done so?

uStepper, 2018-11-03:
Hi BeebBenjamin,

We haven't done this yet, but we do have plans to implement this. 

Best Regards,
Thomas Olsen