uStepper/uStepper #9: home position

natxopedreira, 2017-12-29:
Hello

Im trying to set the position home, but seems reseted if i removed the power and disconnect usb.

Its possible to store permanently a home position?

Thanks!

uStepper, 2018-01-04:
Hi!
Yes, if you use the EEPROM you can save your current position. See here for the Arduino implementation:
https://www.arduino.cc/en/Reference/EEPROM

natxopedreira, 2018-01-04:
Thanks

But think that this will not solve the issue, for example:

I boot up and the stepper get a initial home position, move to 45º and store that (45). On next boot the initial home will not be the same as previously so moving to 45, will not result in the same position as before.

Maybe im missing something here....

Im trying to use the board in mechanical gauge meter so i need to know in each boot where is the zero position.

uStepper, 2018-01-24:
Hi, 
Sorry for the slow reply !
You will need to make an initialization routine, where you manually move the motor shaft to the zero position, and store the raw encoder angle (stepper.encoder.getAngle()) in the EEPROM.

Then everytime the uStepper is powered up after this initialization, you will need to compare the current raw encoder angle against the value stored in EEPROM, and move the motor back to zero position according to the difference. once the uStepper has been moved to zero position, you can call the setHome() function, and everything will be referenced to the zero position.

This assumes that the shaft is not rotated more than one turn during power outage.

I hope this helps!

natxopedreira, 2018-01-25:
I understand, sounds like a good solution

Thanks!!

closed 2018-01-25 by natxopedreira