uStepper/uStepper #18: Limit detection difference calculation

bengineer19, 2018-04-30:
Limit detection seems to be broken due to an incorrect calculation.

Here's the [line in question](https://github.com/uStepper/uStepper/blob/b927a879fccbc5a02967af6d28a8fcf58780b072/src/uStepper.cpp#L1476).

The line fails to find the difference between the two values if, eg `getAngleMoved()` is negative and `lengthMoved()` is positive.

So, if you just `abs()` both of them before taking the `abs()` of the difference or something like that, that would work fine :)

uStepper, 2018-05-01:
Hi Ben,

Thank you for pointing this out ! :)

although in this case, the problem is not in the software calculation, but rather in the orientation of the physical motor connection to the PCB. The sign of getAngleMoved() is dependent of the shafts direction of rotation, e.g. the orientation of the motor connector. if you turn the motor connector 180 degrees, the problem disappears, as getStepsSinceReset() and getAngleMoved() counts in the same direction.

However, there is a flaw in the line you mention (and i will correct this), the step resolution multiplier (0.1125)  is only valid for 16th microstep setting, where it should be different depending on the settings used.

Best Regards,
Thomas Olsen

bengineer19, 2018-05-04:
Thanks Thomas.

If this is the case, then might I suggest that you change the Fig 8 in your [product sheet](http://ustepper.com/product_sheet_revB.pdf), which shows the motor being connected the 'wrong' way. 

Alternatively, have a variable passed at setup() which allows the orientation to be set.

wormyrocks, 2018-07-23:
hi,

would it be possible to get a suggestion on exactly what the right orientation is to connect the motor, then?

with the ego shield/photogrammetry demo i can only get it to successfully limit-find on one axis, and inevitably the moveToAngle command at the end of the limit function forces it past the end of the rail.

i suspect the same issue occurs when PID is enabled on the latest FW version, since for that one the error calculation occurs in detectStall(), but i'm not sure what the best/safest value to change is.

[and it seems like other folks have been having this problem as well](https://forum.arduino.cc/index.php?topic=488822.0)

please let me know, thanks!