uStepper/uStepper #22: S-Line?

npgoalkeeper, 2018-09-05:
If the crowdfunding campaign for the uStepper S line fails, will the Gerber files be published here?

uStepper, 2018-09-06:
Hi, we hope that it will not come to that. But in the case it does, we will release the source files on git when we produce and verify the first batch of either of the boards. I.e. the uStepper S-lite will come relatively soon while the uStepper S could be a while and the S10 a long time. We of course hope people will support our campaign since we put a lot of hours in development of both hardware and software. It's a spare time project but we have to get some revenue to pay insurances, accounting, software licenses and hopefully also have money for further development. If we dont have the money, development halts until we have...

npgoalkeeper, 2018-09-06:
Okay. Thanks for the quick response.

closed 2018-09-06 by npgoalkeeper