uStepper/uStepper #23: Non-commercial license clause hinders potential use

tobbelobb, 2018-09-09:
Hello uStepper devs!

Congrats with the ongoing Kickstarter campaign. I'm lead dev of the Hangprinter, which uses Mechaduinos and Smart Steppers. The uStepper, and the coming new models, look awesome from a technical perspective. It also seems like you intend your product to be open source hardware, which is awesome.

I teach people to build and use Hangprinters. Participants take their self-build machines with them home together with copies of all the relevant source files. Since the tickets to my workshop events cost money, I'm hindered from using your product.

From reading on your campaign site and home page, you seem unaware that "open source" and "non-commercial" are incompatible licensing schemes. Your source can not be open source and non-commercial at the same time. See for example the [open source definition](https://opensource.org/osd) and [CC wiki page explaining NC clause](https://wiki.creativecommons.org/wiki/NonCommercial_interpretation#Choosing_NC_for_your_content) for more.

The NC clause hinders most potential users, including schools, universities, from using your copyrighted material legally. Potential cross-pollination with other projects (open source or not) is also hindered. See for example these sources for more info:
https://community.oscedays.org/t/why-are-non-commercial-licenses-not-open-source/569
http://freedomdefined.org/Licenses/NC

Sorry for preaching if you already knew all this. You're of course free to choose any license you like for your work. I wish you the best of luck with your project regardless!

uStepper, 2018-09-09:
Hi,

We are fully aware of the limitations the NC license gives. We researched it three years ago when launching the first uStepper. We are however, still discussing the licensing question internally. Your comment is taken into account and appreciated.

Best regards, 
Team uStepper

closed 2018-09-09 by uStepper

tobbelobb, 2018-09-09:
Thanks for your quick answer!

It's not really up to you to consider the project to be open source or not. Your work is open source if you're using a license that fulfills the open source definition.

However, the license is between you and your users. If your users get the source along with an open source license, then you're open source. You're not required to give source code or any specific license to non-users (like I currently am) in order to be open source.

So, what you're writing in the last section could sense. If I become a user, and you grant me rights to use it under an open source license, then your work is open source. This new license would include giving me the rights to further spread your work under an open source license.

If you choose to continue to use the term "open source" without fulfilling its definition, you're obfuscating the work of the open source community that I am part of. I recommend terms like hackable, maker-friendly, and free-to-download.

Best regards

fallenartist, 2019-03-14:
I'm a bit confused here. If I purchase the uStepper hardware from your shop, am I free to use it wherever I want, including in a product that generates income for my company, or not?

uStepper, 2019-03-14:
Hi,

As long as you buy the uStepper hardware from us, you are free to do whatever you want with the product, including using it in a commercial product.

I hope this clears up the confusion

fallenartist, 2019-03-14:
It does, thank you.