uStepper/uStepper #25: isStalled() does not work

palmhoej, 2019-10-06:
I am using the Ustepper in an application were i stall it on purpose. But i cannot get the isStalled() function to work. It is not highlighted as library variable in the Arduino IDE (as suppose to moveSteps() which get yellow font when written) but it does compile fine, it just does not work, as it does not return the Serial print in the below code when i stall the motor:

if(stepper.isStalled()==1){
  Serial.println("stepper Stalled");
}

palmhoej, 2019-10-07:
Ok IsStalled is actually working, when i turn of the power it kicks in after a second or so. But when i actually stall the motor it does not come on (I stalled it for 2-5 seconds)... what triggers this function? Is it missing steps or a time variable? And is it possible to set these variables?

uStepper, 2019-10-08:
Hi,

 

What uStepper board are you using?

And have you placed the encoder magnet on the motor? uStepper uses it’s encoder feedback to detect a stall.

 

Best regards,

Mogens

palmhoej, 2019-10-12:
Hi Mogens.

Yes the magnetic encoder has been placed and it is running in the PID mode. If i move it away from it position with force it is returning when i let go. 
Maybe it is because i do not understand the stepper.isStalled() function. What i am trying to achieve is driving a drum for sorting purposes and if it is not in the commanded position after lets say 1000ms (1 sec) or maybe 2000ms i would like an alarm to stop and alert the operator of a possible jam. I thought that was what this function was used for? Are there some parameters i can set for this function, like time or errorthreeshold?

Another thing i am having trouble is tuning the PID loop. I can't see any examples (under the uStepperS example list in the Arduino) on how i can set the P, I & D parameters, when running the stepper.setup(PID,200); code? What commands are used for this? My motor starts oscillation when i command a position, probably due the heavy load of the drum. Also how do i set the threeshold for a reached position? At this point i have 51200 steps/rev and if i am within 100-500 steps i am happy (i do not need precision).

I would also like to dial it down to maybe half steps, again i do not need precision and if its half stepping, the PID loop does not need to work as hard, making it easier to tune. But i have not been able to find any examples on how i set this parameters through the arduino code.

Sorry if this is obvious questions, i have just not been able to find any page with examples were these parameters are used (google search), but maybe you can point me to it? It would be nice to have all the above in the example sketch for the StepperS library.

palmhoej, 2019-10-12:
Oh sorry i forgot to answer your first question:

I have the uStepper S version i bought from your last Kickstarter campaing.

Best regards Kristoffer.

uStepper, 2019-10-12:
Hi,

The name PID for the corrective mode is misleading we can see. Actually the PID mode does hard corrections, i.e. it does not require any parameters other than how many full steps your motor has per revolution (200 for an 1.8deg motor).

I am not quite sure if you need PID at all with the description you give here. So, you really just need to move to motor open loop and verify that It got to the position within 1 second?

For that I would simply just use NORMAL mode and moveToAngle (xx), where xx is of course the desired angle and then use millis in an if statement (so, record millis value before start and compare that with the actual millis) to check if you passed a second, read the encoder value and check if it matches the desired position.

Microstepping is dynamic, so it will automatically go from full step to 1/256th step – that is a feature of the driver chip.

The stepper.isStalled() takes an argument – sensitivity between 0 and 1. Low number is more sensitive so, e.g. stepper.isStalled(0.9). It will return 1 when stall is detected so evaluating it would be correct e.g. if(stepper.isStalled(0.9)) then do something.

We will work on better documentation and think the “PID” function through again, at least the naming.

In the meantime could we continue this support case on email? sales at ustepper.com

Best regards,

Mogens

uStepper, 2019-10-12:
Or actually we do make PID control, it is just on the speed. So, if there’s a position fault we make PID control on the speed but the parameters here are not configurable from any function call or setup for now.

Best regards,

Mogens